punch-mission/punchbowl #307: We can speed up mosaic building

svank, 2024-11-15:
Right now when building mosaics, each input image is reprojected into the full output frame. Each input (or its bounding box) spans maybe a third of the output frame, so we're doing about 3x more WCS calculations than we need to. By reprojecting into a cutout of the output frame (equal to the bounding box of the input image's projection) and then combing those into the full mosaics, we can probably speed up the mosaic building by 3x.

This can be done by first projecting a bunch of points along the edges of the input image into the output frame and finding the min/max coordinates of those points. (Yesterday I wrote code to do that for the helio->celestial reprojections in L3).

`reproject` also has `reproject_and_coadd`, which takes in multiple images and builds a mosiac, and it does this bounding-box calculation automatically. It has a few modes for how to handle the overlap region between images---not sure if there's one that does what PUNCH wants.

(Ultimately, I feel like this is a feature that should be added to `reproject` itself for `reproject_adaptive` et al.)

jmbhughes, 2024-11-18:
Closed in #310

jmbhughes, 2024-11-25:
Reopening because there's more we can do.

PR #326 (2024-11-26): Make mosaics faster w/ bounding boxes
## PR summary

When building trefoil mosaics, this PR finds the bounding box of each input image in the output frame, and only reprojects that region. This saves a lot of work for output pixels that never get filled anyway.

For a clear mosaic, the speed increase is from 1m27s to 60s.

## Test plan

I tested this with these files:
```
/d0/punchsoc/gamera_data/1/CR1/2024/11/25/PUNCH_L1_CR1_20241125155055_v1.fits
/d0/punchsoc/gamera_data/1/CR2/2024/11/25/PUNCH_L1_CR2_20241125155055_v1.fits
/d0/punchsoc/gamera_data/1/CR2/2024/11/25/PUNCH_L1_CR2_20241125155055 [... cut]
closed 2024-11-26 by jmbhughes