punch-mission/punchbowl #428: NFI pylon fun issue / masking?

s0larish, 2025-04-17:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

jmbhughes, 2025-04-23:
@s0larish please provide some more detail on this.

jmbhughes, 2026-01-29:
This has been dealt with along the way and will be revisited with our NFI rework.

closed 2026-01-29 by jmbhughes